OrcaSlicer/OrcaSlicer #11: Show the current filament in use, no matter which AMS is selected.  i.e. Switch focus to current AMS

SwagDad99, 2022-12-01:
Issue:
Right now, when using more than one AMS, if the filament in use is from a different AMS than the one that has 'Focus', or is selected above the filament colors, it shows a 'blank' color in the extruder, and no color in the 'Path' from spool to extruder.  It should show the current path from the spool in use to the extruder.

Resolution:
In the device tab, while using more than one AMS, when the filament begins coming from the alternate AMS, switch focus to that AMS so that the extruder and path show the correct color, rather than keep it blank unless it is manually selected by the user.

SoftFever, 2022-12-22:
The proposal makes perfect sense to me.
Unfortunately, I don't have the time to investigate this issue. 
Hopefully, BL can solve it soon.

closed 2022-12-22 by SoftFever

SwagDad99, 2022-12-27:
Are they, Bambu, aware of the issue?  Do they see what is posted to your GitHub, or does it need cross posted to theirs?

Thank you,

Don

From: SoftFever 
Sent: Thursday, December 22, 2022 8:32 AM
To: SoftFever/BambuStudio-SoftFever 
Cc: SwagDad99 ; Author 
Subject: Re: [SoftFever/BambuStudio-SoftFever] Show the current filament in use, no matter which AMS is selected. i.e. Switch focus to current AMS (Issue #11)

The proposal makes perfect sense to me.
Unfortunately, I don't have the time to investigate this issue.
Hopefully, BL can solve it soon.