OrcaSlicer/OrcaSlicer #9: Add FLSUN V400 printer profile

robanderson, 2022-12-01:
I would like to request a printer profile for the popular FLSUN V400 be added to list.

I noticed that when defining the print bed there is currently no circular option that exists in PrusaSlicer. 

I'm happy to donate $100 via PayPal to this project if anybody can add the V400 to this project.

I currently run 2x X1CC 3x Vorons and 1x V400 and would love to use this slicer for all my printers rather than have to maintain and learn Cura (the default V400 slicer)

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

SoftFever, 2022-12-05:
A bit of underground work is required to support delta printers.
Those printers have their origin(0,0) point at the center of the bed.
V400 will be added once I fix that first.

tokamac, 2022-12-08:
That would be wonderful, as I have both an X1-Carbon and a V400! 
Especially, I tried to print a very demanding model with a very important curvy slope but PrusaSlicer and BambuStudio treat differently the top surface layers and their various settings. Even when tuned equally, they produce different outputs. As a result, the file sliced with BambuStudio or your SoftFever fork works on the X1C but fails to print on the V400 (sliced in PS). 
This may be a printer-related issue. But I feel the slicer is the culprit. This is a very specific case I know.

cpasoft, 2023-03-19:
I'm guessing that's going to be very low on the list of things to develop but, can I ask if there is some kind of deadline for this.

Thanks and congrats for this beautiful piece of software... ;-)

Naninani, 2023-04-04:
@SoftFever, if you add the support for the circular bed shape, I will submit a pull request to add the V400 config files. I have it dialed in, just need the circular bed shape support.

arshish1612, 2023-04-07:
Is there any update on this?

tokamac, 2023-04-07:
Indeed, this request has won the title of the oldest open issue on this Github repo.

SoftFever, 2023-04-09:
Sorry guys. I know it takes a little bit longer than expected.
Anyway, I spent some time investigating delta printers this weekend, and I think I made some progress.
Hopefully, I can release an alpha build for delta printers this week.
G-code looks right to me. 
I don't have any delta printers so I can't test it by myself.
[image]
[image]

tokamac, 2023-04-09:
We will test it :)

Jokiblack, 2023-04-09:
> Sorry guys. I know it takes a little bit longer than expected. Anyway, I spent some time investigating delta printers this weekend, and I think I made some progress. Hopefully, I can release an [...]

Send a beta and we'll start testing with v400.

MrGnarly90, 2023-04-12:
Same i have two heavily modified SRs which makes my circle 265mm and have direct drive. Happy to help test, been WAITING for this! thank you for all your hard work sir!

MrGnarly90, 2023-04-17:
following up on this :) any word on a test version?

MrGnarly90, 2023-04-17:
Thank you for the beta release!

tokamac, 2023-04-17:
Whoohooohoooo 🥳
https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.6.2

SoftFever, 2023-04-17:
Delta printers are supported now.
Feedbacks or suggestions for the default profiles are welcomed.
Also let’s move on the discussion thread🥰
This ticket will be closed soon

robanderson, 2023-04-17:
[image]

closed 2023-04-17 by robanderson

SoftFever, 2023-04-18:
> [image]

@robanderson This is nuts! Thank you so much :)