OrcaSlicer/OrcaSlicer #14: Error when using ./BuildLinux.sh -dsi

WZ9V, 2022-12-03:
When I attempt to build a Linux version using the BuildLinux.sh shell script it errors out with the following messsage

make[2]: *** [CMakeFiles/Makefile2:1017: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:509: src/CMakeFiles/BambuStudio.dir/rule] Error 2
make: *** [Makefile:218: BambuStudio] Error 2

I did a the following sequence according to the instructions
./BuildLinux.sh -u     // all version dependencies checked out
./BuildLinux -dsi      // got above error

System:
  Kernel: 5.15.0-56-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.4.12
    tk: GTK 3.24.33 wm: Mutter dm: LightDM Distro: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy

H4kunaM4tata, 2022-12-08:
I don't think this is a SoftFever issue when not even the official one has a Linux version.
For what matters, I can generate SoftFever AppImage just fine using that same script.

SoftFever, 2022-12-13:
@H4kunaM4tata do you mind sharing your AppImage build script(through PR) so that other Linux users can use it?

-SoftFever

H4kunaM4tata, 2022-12-13:
> @H4kunaM4tata do you mind sharing your AppImage build script(through PR) so that other Linux users can use it?
> 
> -SoftFever

`https://github.com/SoftFever/BambuStudio-SoftFever/blob/SoftFever/BuildLinux.sh` is all I do:

- `BuildLinux.sh -u` (first time only)
- `BuildLinux.sh -dsi `

The last 2 releases has been only the last command, and everything works fine, I have the `AppImage` at the end.
Like I mentioned on Discord, this is not the first time I am compiling stuff on my PC even so `BuildLinux.sh `does download and install packages needed to compile everything, I already had everything needed to compile stuff on Linux.

So yeah, I am not doing anything special other than running the script itself.
I cannot tell why that error, when it comes to compiling stuff, I am just a curious ordinary user haha

I am running Linux Mint Cinnamon 20.4 for what it is worth it.
I cannot remember why but something on Linux Mint 21 kept me back.
Also, I do remember trying Linux Mint Mate and I was forced to go back to Ubuntu, Mate package management is just terrible.

This user is running Cinnamon but a newer version.
When I have free time, I will spin-up a VM to test this Cinnamon 21 out.

closed 2022-12-20 by SoftFever