OrcaSlicer/OrcaSlicer #15: Version 1.3.4 not running

prawnz, 2022-12-04:
**Describe the bug**
Latest release for windows doesn't seem to run in Windows 11. Task manager showed it appearing for a min then shuts down.

**To Reproduce**
Steps to reproduce the behavior:
1. Download from https://github.com/SoftFever/BambuStudio-SoftFever/releases/download/v1.3.4/BambuStudio-SoftFever_V1.3.4.zip
2. Unzip
3. Install runtimes
4. Run Bambu-Studio

**Expected behavior**
Slicer to open.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

SoftFever, 2022-12-05:
Did you install both runtimes? 

> 2. If you have troubles to run the build, you might need to install following runtimes:
> [MicrosoftEdgeWebView2RuntimeInstallerX64](https://github.com/SoftFever/BambuStudio-SoftFever/releases/download/v1.0.10-sf2/MicrosoftEdgeWebView2RuntimeInstallerX64.exe)
> [vcredist2019_x64](https://github.com/SoftFever/BambuStudio-SoftFever/releases/download/v1.0.10-sf2/vcredist2019_x64.exe)

prawnz, 2022-12-05:
>    1. If you have troubles to run the build, you might need to install

Yup both done

prawnz, 2022-12-06:
I found out issues. its due to Windows 11 22H2. I resinstalled to 21H2 and it all worked.. 

Sorry for the inconvenience

closed 2022-12-06 by prawnz