OrcaSlicer/OrcaSlicer #18: inner-outer-inner/infill seems to be creating some unwated effects

H4kunaM4tata, 2022-12-08:
I couldn't understand why my model out of the blue was showing this weird issue. 

[image]

After checking the comment https://discord.com/channels/970163353542606888/982154261897019412/1050199936127479918  I went to check my files and found the only thing that was different.
The body has the standard order of wall while the head has the new one released with 1.3.4

[image]

[image]

I am printing it again with the standard order to confirm.

SoftFever, 2022-12-13:
Interesting.
What kind of gap fills caused this issue? (those long big gap fills or the tiny gap fills?)

-SoftFever

H4kunaM4tata, 2022-12-13:
> Interesting. What kind of gap fills caused this issue? (those long big gap fills or the tiny gap fills?)
> 
> -SoftFever

If I understood your question, this is my setup for all the profiles:

[image]

I haven't done any further tests to make sure I didn't do anything stupid.
But it is fair to mention that another user printing CF had some weird experience with his model showing weird lines not present with the default setting.

I did compare prints' 3MF files with and without the changes, all the ones like Joker, all had the new order while the ones that printed fine, they all had the default order.

SoftFever, 2022-12-18:
Btw, how many `wall loops` did you use?

H4kunaM4tata, 2022-12-18:
> Btw, how many `wall loops` did you use?
> 

2 walls and 10% infil, this is my default printing settings for anything non functional or heat resistant.

I need to do further printing tests, I've been busy trying to compile the latest release. No SF related, something changed so some Linux users with me included are "stuck" with the previous slice version.

SoftFever, 2022-12-19:
> 

This explains.
In the case of two walls, this sandwich mode will degrade to the traditional `outer wall first`.
The outer walls will be printed directly followed by long fast travels, there are chances to cause such artifacts.
This is exactly the reason I implement this new strategy.

-SoftFever

H4kunaM4tata, 2022-12-19:
I see so that means I must print with 3 walls always?!

SoftFever, 2022-12-19:
That's right.

SoftFever, 2022-12-19:
I will close this issue as this is not a sandwich mode issue anymore.
Free feel to create a new issue if you found the issue persists with 3+ wall loops.

-SoftFever

closed 2022-12-19 by SoftFever

H4kunaM4tata, 2022-12-19:
It seems to be fine so far, I just printer a cube at 0.32
Will keep monitoring things closer.
It is fair to close it.
Thank you for checking this out.

[image]

SoftFever, 2022-12-19:
> It seems to be fine so far, I just printer a cube at 0.32 Will keep monitoring things closer. It is fair to close it. Thank you for checking this out.
> 
> [image]

Cool! Keep us posted with your new testing.
I've received very positive feedback for the sandwich mode.
The following image is provided by a user: 
Sandwich vs inner-outer-infill.
[image]

Although I don't really recommend printing such models with sandwich mode.
It's meant for those engineering parts. But still, I'm very glad to see it works well for this kind of model too.

-SoftFever

beaverdude, 2025-05-20:
Hello! I bumped into this issue searching for explanation of a very surprising behavior when using sandwich wall mode. 

It looks like a border case when the 3rd line in sandwich is actually a little gap to be filled. After that gap the nozzle jumps to external line somewhat underprimed. 
@Funkton and @SoftFever  mentioned gaps fill in this thread

[image]

[image]

[sandwich_artefact_PLA_7m28s.zip](https://github.com/user-attachments/files/20351688/sandwich_artefact_PLA_7m28s.zip)