OrcaSlicer/OrcaSlicer #22: Klipper outputs error in regards to G17 after update

Fl0ken, 2022-12-12:
**Describe the bug**
Added the G2/G3 arcs and the dummy command to printer.cfg and used the firmware and arcs without any issues. Updated Klipper to v0.11.0-13-g201f715b and now there is an error that G17 is already registered. Commenting the G17 command out solves this. As it's recommended to have this, any solution to fix this? use "rename_existing"?

**To Reproduce**
Steps to reproduce the behavior:
1. Update to Klipper v0.11.0-13-g201f715b
2. Get error from klipper

**Expected behavior**
No issues when updating klipper.

**Screenshots**

SoftFever, 2022-12-13:
You can simply delete the dummy G17 macro since the latest Klipper has supported this command.
I will update the readme later.

-SoftFever

Fl0ken, 2022-12-13:
Unfortunately this does not fix it. I now get an error when the gcode has G2/G3 arcs: "G2/G3 requires IJ, IK or JK parameters"

I can't pinpoint exactly why it works for some files (a coin worked without issues) but a larger print with text did not.

SoftFever, 2022-12-13:
This very latest Klipper update breaks the ARC support: 
[image]

https://github.com/Klipper3d/klipper/commit/0c5c87d7c08c2b8aa2f248d81dd1003c5c7d5a7c#diff-4f1c41066bdb7bfe33f5c832f650b43f1bc088ce7a35edd116d2e5a50bcb39a7

I will see what needs to be done to support that.

As a workaround, for now, you can either roll back the Klipper to the previous version. 
Or turn off the arc support in the slicer. (I'm not sure if the `arc fitting` or the `spiral lift` caused this issue, I suspect the latter)

-SoftFever

SoftFever, 2022-12-14:
@Fl0ken I can't reproduce the issue.
Can you share the project?

-SoftFever

Fl0ken, 2022-12-14:
Here's a .3mf and gcode file which produced the error. Happened on the first layer already. [ARC_project.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10228284/ARC_project.zip)

SoftFever, 2022-12-14:
After investigation, I believe it's a newly introduced bug in the latest Klipper.
I saw someone else has raised an issue and PR already: https://github.com/Klipper3d/klipper/pull/5937

-SoftFever

tehmaze, 2022-12-14:
With #5939 merged this should be resolved, please confirm.

Fl0ken, 2022-12-15:
That seems to have resolved it. Thanks!

closed 2022-12-15 by Fl0ken