OrcaSlicer/OrcaSlicer #26: Cannot build whith lastest commits

GiaPag, 2022-12-13:
**Describe the bug**
Hi, there are some errors in compilation whit latest commits

`/Users/giapag/BambuStudio-SoftFever/src/hints/HintsToPot.cpp:12:29: error: 
      expected ';' after expression
        boost::filesystem::ofstream file(std::move(path), std::ios_base::app);
                                   ^
                                   ;
`

`/Users/giapag/BambuStudio-SoftFever/src/hints/HintsToPot.cpp:12:21: error: 
      no member named 'ofstream' in namespace 'boost::filesystem'
        boost::filesystem::ofstream file(std::move(path), std::ios_base::app);
        ~~~~~~~~~~~~~~~~~~~^`

`/Users/giapag/BambuStudio-SoftFever/src/hints/HintsToPot.cpp:12:30: error: 
      use of undeclared identifier 'file'
        boost::filesystem::ofstream file(std::move(path), std::ios_base::app);
                                    ^`

`/Users/giapag/BambuStudio-SoftFever/src/hints/HintsToPot.cpp:20:3: error: 
      use of undeclared identifier 'file'
                file << "\n#: resources/data/hints.ini: ["<< element.fir...
                ^`

`/Users/giapag/BambuStudio-SoftFever/src/hints/HintsToPot.cpp:22:2: error: 
      use of undeclared identifier 'file'
        file.close();
`

**Desktop (please complete the following information):**
 - OS:macOS
 - Version 13 (Ventura)

SoftFever, 2022-12-13:
I'm able to build it without issues on Monterey(12.6).
Looks like an issue specific to Ventura.

-SoftFever

hliebscher, 2022-12-14:
I'm also able to build it without issues on Monterey 
I can compile the bug fix version for you

SoftFever, 2022-12-14:
Thanks for the confirmation.
Looks like a false alarm.
I will close this issue then.

closed 2022-12-14 by SoftFever

GiaPag, 2022-12-14:
> I'm also able to build it without issues on Monterey
> I can compile the bug fix version for you

Thank you