OrcaSlicer/OrcaSlicer #28: Specific fan at layer

GiaPag, 2022-12-14:
It would be nice to be able to decide the fan speed in certain layers.
I've only seen such an option on IdeaMaker
this is an example
[image]
This function is very useful in order not to have a forced fan at a higher speed for the entire print. In difficult pieces it helps a lot for overhangs and also avoids making a piece more fragile having avoided a bigger fan for the whole print.
By this I am referring to technical sections where, for example, in some areas we find difficult overhangs but we have few walls

SoftFever, 2022-12-14:
I would argue that with the gradient overhang speed we have in the slicer, this is not needed anymore.
[image]

GiaPag, 2022-12-14:
Hi, yes but it's not the same thing instead to the possibility to specific fan speed at layer

SoftFever, 2022-12-22:
Given the limited time I can spend on this fork, we won't have this feature in the near future.
Unless there someone volunteered to work on this.

I am closing it.

closed 2022-12-22 by SoftFever

GiaPag, 2022-12-22:
I see, thanks anyway