OrcaSlicer/OrcaSlicer #29: 3 app crashes in 2 days

SwagDad99, 2022-12-17:
Leave the app running, streaming video.  Work another model in the prepare tab.  Leave focus of the app, going to Discord, etc.  Some time later, notice that the app is no longer running.  R-launch the app, and it asks me if I want to restore the project in progress...

**3mf File for This Bug**
different projects

**To Reproduce**
Described above

**Expected behavior**
App is there when I want to return to it.

**Screenshots**
None

**Desktop (please complete the following information):**
 - Windows 11 Home
 - 32 Gig ram

SoftFever, 2022-12-18:
Let me try to leave it opened the whole night tonight and see if I can repro it(while watching WorldCup Final ;)

SoftFever, 2022-12-21:
I think I found the cause.
BambuSource.dll(the plugin) causes the crash.
The crash only happens when I have the camera stream on in the device tab.

```
Faulting application name: bambu-studio.exe, version: 1.0.0.0, time stamp: 0x63a13407
Faulting module name: BambuSource.dll_unloaded, version: 0.0.0.0, time stamp: 0x6396ca02
Exception code: 0xc0000005
Fault offset: 0x0000000000084ce1
Faulting process ID: 0x0x2DFC
Faulting application start time: 0x0x1D914E7D8E2E739
Faulting application path: C:\Codes\BambuStudio-SoftFever\build\BambuStudio-SoftFever_V1.4.0\bambu-studio.exe
Faulting module path: BambuSource.dll
Report ID: 8e69814f-39e1-478e-bcd6-ceb18420358f
Faulting package full name: 
Faulting package-relative application ID: 
```