OrcaSlicer/OrcaSlicer #30: network plugin fails to install

hliebscher, 2022-12-18:
After Upgrade to 1.4 beta  (incl the last merge)

The network plugin fails to install when opening the program. 

It worked OK with v1.3.4 and the official Bambu Studio 1.4
at MacOS 12.1

deleting the plugins directory is the same ...

SoftFever, 2022-12-18:
Does it fix the problem if you rename(don't delete it contains all your configs) the `/Users/<yourname>/Library/Application Support/BambuStudio-SoftFever` folder?

SoftFever, 2022-12-18:
It works fine on my Mac(12.6.1)

hliebscher, 2022-12-18:
i found the bug

[download_plugin]: get version of settings's type, "slicer/plugins/cloud"
2022-12-18 11:50:36.409071[Thread 0x000000029a507000]:[download_plugin 1]: get type slicer/plugins/cloud, version 1.4.0.15, url https://public-cdn.bambulab.com/upgrade/studio/plugins/01.04.00.15/mac_01.04.00.15.zip
2022-12-18 11:50:36.409140[Thread 0x000000029a507000]:UpgradeNetworkJob: percent = 5msg = Downloading
2022-12-18 11:50:36.409371[Thread 0x000000029a507000]:[download_plugin] get_url = https://public-cdn.bambulab.com/upgrade/studio/plugins/01.04.00.15/mac_01.04.00.15.zip
2022-12-18 11:50:36.983294[Thread 0x000000029a507000]:UpgradeNetworkJob: percent = 0msg = Download failed
2022-12-18 11:50:36.983460[Thread 0x000000029a507000]:[download_plugin 2] on_error: curl:**SSL connect error:
SSL peer handshake failed, the server most likely requires a client certificate to connect**
[Error 35], body = 

manually download works

https://public-cdn.bambulab.com/upgrade/studio/plugins/01.04.00.15/mac_01.04.00.15.zip

SoftFever, 2022-12-18:
Cool.
Thanks for sharing the investigation.
I'm closing this issue for now.
We can revisit it if there are more reports in the future.

-SoftFever

closed 2022-12-18 by SoftFever

GiaPag, 2022-12-18:
Hi. 
I have the same bug on macOS Ventura with latest 1.4 beta 3
```* * Connection #0 to host api.bambulab.com left intact
*   Trying 99.86.91.19:443...
* Connected to public-cdn.bambulab.com (99.86.91.19) port 443 (#0)
* ALPN, offering http/1.1
* WARNING: disabling hostname validation also disables SNI.
* SSL peer handshake failed, the server most likely requires a client certificate to connect
* Closing connection 0
```

H4kunaM4tata, 2022-12-19:
This isn't a SF problem.
On Linux, I can download the file but there is a problem with the file itself.
This is a BL problem to deal with.

`BambuStudio-SoftFever/plugins/libbambu_networking.so: undefined symbol: __libc_single_threaded`

hliebscher, 2022-12-19:
It works after another compile with mac without any problems...
it can be a problem of the Loadbalancer from CDN from Bambulab