OrcaSlicer/OrcaSlicer #31: Feature Request for Solid Infill per specific layer

brycehoeper, 2022-12-18:
**Is your feature request related to a problem? Please describe.**
PrusaSlicer has an option in the Print Settings > Infill > Advanced section called "Solid infill every: ___" that allows a 100% solid infill per specific layer.  I need this to be able to print the prints that I make.

**Describe the solution you'd like**
Would you be able to add this "Solid Infill every:____" to SoftFever?

**Describe alternatives you've considered**
I don't think custom gcode would allow me to add this manually per feedback I've been given in the Bambu Lab discord

**Additional context**
Thank you

Learoy1, 2022-12-28:
I'd love this feature too

closed 2023-11-25 by github-actions[bot]

liftbag, 2024-01-11:
Is it possible to reopen this feature request? On some occasions it would have been useful to me.