OrcaSlicer/OrcaSlicer #32: [FEATURE] Use the "filename format" when exporting the 3MF

H4kunaM4tata, 2022-12-18:
Under normal circumstances, when exporting the g-code, you can have this beautiful file name

[image]

Within a folder with dozens of files, that allows you to easily identify the models, layer height, which filament, how many grams of filament and how many hours that model will take.

[image]

But when exporting the same project as 3MF, since you haven't saved anything, the project has no name so it generates this useless file name

[image]

And as consequence, you have all these useless files that you cannot tell if you sliced with PLA, PLA+, PETG, ASA, you have no idea how long and if you have enough filament

[image]

It would be a dream to be able to save the slicing as:

[image]

Thank you

zeroemso, 2022-12-20:
+1 buddy i was on the way to open a request too. thx.

H4kunaM4tata, 2022-12-20:
> +1 buddy i was on the way to open a request too. thx.

I was supposed to open this request like ages ago but I kept forgetting.
It is becoming impossible to identify the right files now, too much printing haha

zeroemso, 2022-12-20:
if you came from klipper it is very hard without a mainsail "file section" now i have to name my files different from vorons + no Folders on SDCard is a real pita.

SoftFever, 2022-12-24:
Done. 
New changes will be included in the next update

SoftFever, 2022-12-24:
[image]

closed 2022-12-24 by SoftFever

H4kunaM4tata, 2022-12-24:
> [image]
> 

Yes 🙌
Are you able to add 2 more info??

The layer height and the grams of filament used like the screenshot I shared???
Then you have all the important information on the filename.

Thank you so much for looking into this.

SoftFever, 2022-12-24:
> > [image]
> 
> Yes 🙌 Are you able to add 2 more info??
> 
> The layer height and the grams of filament used like the screenshot I shared??? Then you have all the important information on the filename.
> 
> Thank you so much for looking into this.

The name is configurable just like the gcode file. You can customise it the way you liked

SoftFever, 2022-12-24:
> PrusaSlicer here.
> 
> I like to don't depend on extra stuff as less as possible so this format:
> 
> `{input_filename_base}_{layer_height}mm_{filament_type[0]}_{print_time}_{round(total_weight)}g.gcode`
> 
> Gives me a gorgeous filename: `Mandalorian_toothbrush_holder_0.2mm_PLA_1h56m_16g.gcode `
> 
> It is easy to find stuff, I don't need to load the file anywhere to see anything, its filename tells me everything I need to know. On your cause I would have:
> 
> * `X1C_Mandalorian_toothbrush_holder_0.2mm_PLA_1h56m_16g.gcode `
> * `Voron_Mandalorian_toothbrush_holder_0.2mm_PLA_1h56m_16g.gcode `
> 
> I am hoping this is something "easy" to implement. My OCD is going wild hahaha

Have you tried this file name format for the gcode file name? It should also work in bs sf version

H4kunaM4tata, 2022-12-25:
@SoftFever yes, I've tried, copy/paste from PrusaSlicer and it works just fine ONLY if you are exporting as g-code.

I believe is File > Export > Export to g-code

But it doesn't work exporting as 3MF, top right corner. 
That becomes a pain if you export a lot of models having to do all that work and the printer won't load thumbnail so it isn't practical.

If exporting as 3MF it also reads the g-code field you introduced on SoftFever, that would be perfect.

Thank you and Marry Christmas.

SoftFever, 2022-12-25:
> @SoftFever yes, I've tried, copy/paste from PrusaSlicer and it works just fine ONLY if you are exporting as g-code.
>
> I believe is File > Export > Export to g-code
>
> But it doesn't work exporting as [...]

> If exporting as 3MF it also reads the g-code field you introduced on SoftFever, that would be perfect.
> 

Yes, this is the case.

Thanks, merry Christmas and happy New Year!