OrcaSlicer/OrcaSlicer #35: Last Mac Release is corrupt

christianjou, 2022-12-20:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

christianjou, 2022-12-20:
when i try to install the program, i get the message that the file is damaged.

Version: Mac/Intel

SoftFever, 2022-12-20:
> when i try to install the program, i get the message that the file is damaged.
> 
> Version: Mac/Intel OS MONTEREY

Please check this; https://github.com/SoftFever/BambuStudio-SoftFever#1-for-apple-m1-users-please-take-a-look-at-this-article-about-how-to-run-unsigned-applications-on-your-machine-or-better-you-can-build-it-from-the-source-codes-if-you-want

christianjou, 2022-12-20:
I do not have a M1 but an Intel.

hliebscher, 2022-12-20:
Both versions  for MAC 1.4 for M1 and intel are working

https://support.apple.com/de-de/guide/mac-help/mh40616/mac

closed 2022-12-20 by SoftFever