OrcaSlicer/OrcaSlicer #36: Failed to Getting Network Plugin

christianjou, 2022-12-20:
Once it is launch, no way to connect to have the network plugin
So i can not connect to my account.

[image]

Mac OS INTEL/MONTEREY

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

hliebscher, 2022-12-20:
we have a discussion link

https://github.com/SoftFever/BambuStudio-SoftFever/discussions/33

pls wait some hours and than it works...
you can use the Studio without network pluging for non Bambu printer
sending of your GCode to non Babmu printer works

SoftFever, 2022-12-20:
Close as duplicated with #33 .

closed 2022-12-20 by SoftFever