OrcaSlicer/OrcaSlicer #37: variables first_layer_print_min[0] with wrong values

hliebscher, 2022-12-20:
I use M557 for define Mesh Area

M557 X{first_layer_print_min[0]}:{first_layer_print_max[0]} Y{first_layer_print_min[1]}:{first_layer_print_max[1]} S25

in the New build i get wrong values in GCODE back

**M557 X804.147:854.147 Y-531.917:-481.917 S25**

the part size is 50x50 mm 

with the BambuSstudio org and 1.3.4  is this OK. 
 **M557 X102.314:157.686 Y102.314:157.686 S25**

SoftFever, 2022-12-20:
It works fine for me.
Can you try the test project attached?
[image]
[mesh_area_test.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10269295/mesh_area_test.zip)

hliebscher, 2022-12-20:
I found the Bug !!!!
 it works only with **one plate**
it you place some parts to second place the slicer take x0 y0 from the first plate!

hliebscher, 2022-12-20:
one moment

hliebscher, 2022-12-20:
yes the same bug for alternative printers 
bambu have no button to export gcode

hliebscher, 2022-12-20:
it is the same bug

;===== bed leveling ==================================
M1002 judge_flag g29_before_print_flag
M622 J1

    M1002 gcode_claim_action : 1
    G29 A X353.451 Y85.9557 I50 J50
    M400
    M500 ; save cali data

M623
;===== bed leveling end ================================

hliebscher, 2023-01-03:
BAMBULAB

We have fixed it in internal version and will be included in next release.
Thank you very much for reporting issue to keep software improved!

closed 2023-01-03 by hliebscher

SoftFever, 2023-01-22:
fixed

hliebscher, 2023-01-22:
@SoftFever THANKS !!!!!!!!