OrcaSlicer/OrcaSlicer #42: SF takes longer than BS to print

H4kunaM4tata, 2022-12-22:
I am investigating an annoying problem where the printer no longer pauses, yep, it just keeps going.
While I am still testing to identify if this is a BS issue with the latest releases of it this is a SF issue, I noticed that SF takes longer than BS

LEFT: BS
RIGHT: SF

1. same model
2. same printer profile
3. same filament
4. same layer height
5. same everything, BS uses a shortcut to SF `/user` folder so they both will always have the same profiles

[image]

Bigger prints are even worse, over 1h difference.
4h30min against almost 6h with SF

[image]

Arachne which is my to go wall generator improves it but still far behind

[image]

hliebscher, 2022-12-22:
witch OS ?   
on the right side the SF version takes more filament... longer Print time
how long ist the real printing time  BS and SF ?
the most calculated time diff is at the gap infill at the 2. and 3. pictures

H4kunaM4tata, 2022-12-22:
@hliebscher I do not have AMS so those times are 99% accurate as usual.
The biggest time change from SF to the print was like 2-3minutes.

It is worth to mention that I have been using SF since release one.
I am on Linux and I have been having problems to even compile BambuStudio as can follow here: https://github.com/bambulab/BambuStudio/issues/12
That is now solved and as I am investigating what I believe is a firmware problem where the printer does not pause anymore affecting me and other users, I had both slicers side by side. That is when I noticed the time difference

I have no experience with g-code, I have attached both files.

[black_panther_soft_fever.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10283180/black_panther_soft_fever.zip)
[black_panther_bambu_studio.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10283183/black_panther_bambu_studio.zip)

Thank you for looking into this

H4kunaM4tata, 2022-12-22:
EDIT: For what matters, both slicers did not pause the printer, it kept going.
This feature used to work prior to the firmware update.

H4kunaM4tata, 2022-12-22:
@hliebscher this is user error, my apologies.
BambuStusio was set to 2 walls only.
BS It is still half hour faster than SF which I believe it is down to the speeds.
Since BambuStudio doesn't display all the speeds, I believe it has some speed higher than SF.

[image]

I am closing this.
I am sorry for the ooopz.

closed 2022-12-22 by H4kunaM4tata

SoftFever, 2022-12-30:
> @hliebscher this is user error, my apologies. BambuStusio was set to 2 walls only and I somehow missed it. BS It is still half hour faster than SF which I believe it is down to the speeds. Since BambuStudio doesn't display all the speeds, I believe it has some speed higher than SF.
> 
> [image]
> 
> I am closing this. I am sorry for the ooopz.

If you look at those time breakdowns, the only difference is coming from `gap infill`.
I suspect different perimeter width was used hence different amount of gaps were created.