OrcaSlicer/OrcaSlicer #50: Option to Change Solid Infill Pattern

djngnear, 2022-12-24:
It would be great to port over Superslicer's solid infill pattern settings option. And also add a skirt height option. I like using a skirt but when it's only 1 layer tall, it can be tough to remove at times. I usually make mine 3 layers tall.

SoftFever, 2022-12-24:
The skirt height has been added in the dev branch. #20 

Please add more details and use scenarios for the feature you are asking for.
This shall help me or other potential contributors to evaluate it.

-SoftFever

ShryuKt, 2022-12-26:
I don't know if that's what @djngnear  talking about, but one option I like on superslicer is the "Not Connected" infill
[image]
with an infill_overlap of 25%
[image]
this is one (or the only) option that prevents me from switching to bambu studio full time

djngnear, 2022-12-31:
Yeah I was just talking about changing the pattern of the solid infill

[image]

closed 2023-11-25 by github-actions[bot]