OrcaSlicer/OrcaSlicer #51: Compare Profiles

Ibot-11, 2022-12-24:
**Is your feature request related to a problem? Please describe.**
Comparing profiles is pretty annoying atm.

**Describe the solution you'd like**
A comparing tool like prusaslicer or superslicer have.
That would be really nice.

[image]

hliebscher, 2022-12-25:
it is working
[image]

Ibot-11, 2022-12-25:
So is this already in the slicer or will this be added soon?
Maybe i couldn't find this anywhere.

SoftFever, 2022-12-25:
It's already there.
This feature comes from the upstream.
[image]

closed 2022-12-25 by SoftFever

Ibot-11, 2022-12-25:
ok, I found the issue. The language was set to german. And somehow the button disappears then. When I change to English, it's there.
[image]