OrcaSlicer/OrcaSlicer #58: Filament Preset Overrides

DocHogan, 2022-12-25:
Bambu essentially laughed at me when I requested this (https://github.com/bambulab/BambuStudio/issues/522), but I feel it's *essential* for a multi-material system, and is one area that Slic3r-derived slicers are a poor choice for multi-material (what was Prusa thinking, why would he not have implemented this?)

*******************

Feature request:

Add following setting to Filament Preset->Overrides:

Quality/First Layer Height
Quality/Elephant Foot Compensation
Quality/Bottom Surface Flow Ratio

Speed/First Layer
Speed/First Layer Infill
Speed/Outer Wall

SoftFever, 2022-12-30:
A concern of such settings is that in case of multiple filaments are used in one print, and they conflict with each other, it will make the logic and special handling codes very complex - there are already tons of special handling logics there ;(

closed 2023-11-25 by github-actions[bot]

brunetton, 2024-07-09:
> A concern of such settings is that in case of multiple filaments are used in one print, and they conflict with each other, it will make the logic and special handling codes very complex - there are already tons of special handling logics there ;(

Why not in a first implementation ignoring those overrides ? Preventing the user: "Those settings are **ignored** when printing using multiple filaments in the same print"

I really feel the need to be able to store printing speeds with filament settings. I just ruined a print half an hour ago because I forgot to adapt the printing settings to a particular filament I used today. I always have to remember that this filament must be printed very ssslllowwwwww, especially for the first layer. 
I would love OrcaSlicer to remember this for me ! And set the printing speeds when I select this filament :heart_eyes: