OrcaSlicer/OrcaSlicer #61: Small problem with Klipper Flavor

ShryuKt, 2022-12-26:
Hi,

For me there is a little problem when using Klipper Flavor and a full configurable PRINT_START macro.

This is my Machine start G-Code"
```
M190 S0 ; set bed temperature and wait for it to be reached
M104 S0 ; set nozzle temperature
M109 S0 ; set nozzle temperature and wait for it to be reached
PRINT_START     BED_TEMP=[bed_temperature_initial_layer_single]     EXTRUDER_TEMP=[nozzle_temperature_initial_layer]     Z_ADJUST=0     SOAK=0     CHAMBER=[chamber_temperature]     CHAMBER_MAXTIME=0     MATERIAL=[filament_type]     SIZE={first_layer_print_min[0]}_{first_layer_print_min[1]}_{first_layer_print_max[0]}_{first_layer_print_max[1]}
```

And the result in the generated G-Code
```
M190 S0 ; set bed temperature and wait for it to be reached
M104 S0 ; set nozzle temperature
M109 S0 ; set nozzle temperature and wait for it to be reached
PRINT_START     BED_TEMP=110     EXTRUDER_TEMP=240     Z_ADJUST=0     SOAK=0     CHAMBER=40     CHAMBER_MAXTIME=0     MATERIAL=ABS     SIZE=140.275_163.275_199.725_196.725
M190 S110 ; set bed temperature and wait for it to be reached
M104 S240 ; set nozzle temperature
```

Like you can see, the "M190" and "M104" command are repeated at the end even if I defined them in my Start G-Code.
I don't know if it's on purpose, but I think this is an oversight.

More weird, if I remove the M109 command form the Start G-Code, the M190 is remove from the generated G-Code, but the "M104 S240" remains.
Same thing if I remove the "M190"
Some exemples
```
;TYPE:Custom
;M190 S0 ; set bed temperature and wait for it to be reached
;M104 S0 ; set nozzle temperature
M109 S0 ; set nozzle temperature and wait for it to be reached
PRINT_START     BED_TEMP=110     EXTRUDER_TEMP=240     Z_ADJUST=0     SOAK=0     CHAMBER=40     CHAMBER_MAXTIME=0     MATERIAL=ABS     SIZE=140.275_163.275_199.725_196.725
M104 S240 ; set nozzle temperature
```
```
;TYPE:Custom
;M190 S0 ; set bed temperature and wait for it to be reached
;M104 S0 ; set nozzle temperature
;M109 S0 ; set nozzle temperature and wait for it to be reached
PRINT_START     BED_TEMP=110     EXTRUDER_TEMP=240     Z_ADJUST=0     SOAK=0     CHAMBER=40     CHAMBER_MAXTIME=0     MATERIAL=ABS     SIZE=140.275_163.275_199.725_196.725
M104 S240 ; set nozzle temperature
M109 S240 ; set nozzle temperature and wait for it to be reached
```
```
;TYPE:Custom
M190 S0 ; set bed temperature and wait for it to be reached
M104 S0 ; set nozzle temperature
;M109 S0 ; set nozzle temperature and wait for it to be reached
PRINT_START     BED_TEMP=110     EXTRUDER_TEMP=240     Z_ADJUST=0     SOAK=0     CHAMBER=40     CHAMBER_MAXTIME=0     MATERIAL=ABS     SIZE=140.275_163.275_199.725_196.725
M104 S240 ; set nozzle temperature
```

Thanks
--Shryu--

SoftFever, 2022-12-27:
Hi @ShryuKt 

Yes, it's a safety feature.
I added these two additional instructions after receiving many complaints from inexperienced users, like "why it prints without heating my nozzle/bed." :)

These two instructions are inserted after the user's Start G-Code so it shouldn't interfere with any extra logic/workflow in `print_start` macro.

-SoftFever

ShryuKt, 2022-12-27:
Ok thank for your response, and you are right, the instructions does not interfere.
I just wonder why the the M190 disappear if I add M190 or M109 in the start G-Code.

Just for the record, Superslicer has a similar behavior, but when the instructions exist in the "Machine start G-Code" they do not repeat, this is why I put them at the top with "S0"

Thank
--Shryu--

closed 2022-12-28 by SoftFever

ApolloBian, 2023-02-27:
@SoftFever I would suggest adding option to remove the M109/M190 command. SuperSlicer implements it that when user puts `M109 S0` and `M190 S0` in the custom gcode, the slicer will not add extra M109/M190 command. This comes in handy because I use a purge blob and fast nozzle movement to clean up the nozzle. However, if M109/M190 is added after PRINT_START gcode, the machine will occasionally stop and wait for temperature to stablize, which causes filament to ooze out of the nozzle.

ApolloBian, 2023-02-27:
Also for those who have stumbled upon this issue, a solution is to create a python postprocess cript, e.g.

```python
#!/usr/bin/env python3
import sys
import re

filename = sys.argv[1]
with open(filename) as f:
    content = f.read()

pattern ='; the added M109[^\n]*\n((M190|M109|M104)[^\n]*\n)+'
content = re.sub(pattern, '', content, flags=re.S)

with open(filename, 'w') as f:    
    f.write(content)
```

and add the following content to the end of custom start gcode (to help the script locate the added M109/M190 gcode position):
```
; the added M109/M190 gcode is removed with script
```