OrcaSlicer/OrcaSlicer #62: Unable to load Bambulab network plugin.  Softfever 1.4.0 on MacOS 12.6.1

Tonyknz, 2022-12-27:
**Describe the bug**
Bambuslicer application installs and then advises Network plugin not found.

Click to install network plug-in starts the process and then fails with error message "Failed to download the plug-in. Please check your firewall settings and vpn software, check and retry.

As a test on the same machine, an installation of Bambuslicer v1.4.1.52 installed the network plug-in without any problem.

mrmccain9, 2022-12-27:
> **Describe the bug** Bambuslicer application installs and then advises Network plugin not found.
>
> Click to install network plug-in starts the process and then fails with error message "Failed to [...]

Same issue here on Softfever 1.4.0, but on MacOS 11.7

SoftFever, 2022-12-28:
I have no clue why it fails on Mac Intel at the moment.
There is a discussion on it #33 
A workaround for it is to manually download it.

Tonyknz, 2022-12-28:
Manual download of the plug-in following instructions in #33 works.

closed 2022-12-29 by SoftFever