OrcaSlicer/OrcaSlicer #63: Only first 9 of the 18 projects listed in 'Recent Projects' open

SwagDad99, 2022-12-27:
**Describe the bug**
When you click on 'File/Recent Projects', 18 most recent projects are listed.  Clicking on 10-18 in the list do nothing.  The first 9 open.

**3mf File for This Bug**
None

**To Reproduce**
Steps to reproduce the behavior:
1. Have 18 recent projects listed in the 'Recent Projects' list under the 'File' menu.  I am not sure it behaves the same with less recent projects
2. Click on any project in the list, between numbers 10 and 18
3. The screen just returns to the Prepare tab

**Expected behavior**
The selected project should open

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

SoftFever, 2022-12-30:
Works fine on my machine?
[image]

SwagDad99, 2022-12-30:
From the home screen mine works fine.  If you open a project, and then go to the ‘File’, ‘Recent Projects’ menu option, it fails from there.

Don

From: SoftFever 
Sent: Friday, December 30, 2022 2:15 AM
To: SoftFever/BambuStudio-SoftFever 
Cc: SwagDad99 ; Author 
Subject: Re: [SoftFever/BambuStudio-SoftFever] Only first 9 of the 18 projects listed in 'Recent Projects' open (Issue #63)

Works fine on my machine?

SoftFever, 2023-01-04:
I'm not able to repro it.
Can you provide more info?

closed 2023-11-25 by github-actions[bot]