OrcaSlicer/OrcaSlicer #67: Support for RatRig Printers

erebus04, 2022-12-27:
I am currently creating profiles for the RatRig printers.
For the V-Core 3 and the V-Minion.
Is there interest here to integrate this into the standard?

I have attached some screenshots.

[image]
[image]
[image]
[image]
[image]

hliebscher, 2022-12-27:
I can Test an integrate via PR 
I have VCore 3.1 with Klipper und RRF

erebus04, 2022-12-27:
Greate, when I have everything created, I make a PR

SoftFever, 2022-12-28:
> I am currently creating profiles for the RatRig printers. For the V-Core 3 and the V-Minion. Is there interest here to integrate this into the standard? @SoftFever
> 
> I have attached some screenshots.
> 
> [image] [image] [image] [image] [image]

Absolutely welcomed :)

erebus04, 2022-12-28:
Pull Request are created

SoftFever, 2022-12-29:
Merged.
Thanks, @erebus04

closed 2022-12-29 by SoftFever