OrcaSlicer/OrcaSlicer #74: auto sync user presets for 3rd party printer profiles

bavery79, 2022-12-29:
Got a bit excited about Bambu's ability to sync printer/filament/process data from Bambu cloud, but this either doesn't appear to work properly, or maybe some issue with the way 3rd party printer profiles are handled with this SoftFever fork?  Either way, would love to see the profile sync work for non-Bambu printers.

SoftFever, 2022-12-30:
The syncing feature worked well until Bambu made changes in their cloud since 1.4.
It looks like their cloud checks the setting id and refuses the request if the id is not a BBL printer.
While I think I can hack to make it work now, it's indeed better to have the capability to sync with Google/Apple drive in the long run.

SoftFever, 2022-12-30:
fixed

closed 2022-12-30 by SoftFever