OrcaSlicer/OrcaSlicer #75: Application Crash when adding text - Mac 12.6.1 / Bambulab Softfever 1.4.0

Tonyknz, 2022-12-29:
**Describe the bug**
Application crashes when using the add text function.

**To Reproduce**
Steps to reproduce the behavior:
Create an object within the slicer
Select the Text icon
Enter any text
Click ADD
Application crashes.

**Desktop (please complete the following information):**
 - OS: Mac OS / Intel
 - Version 12.6.1 (Monterey)

Partial Crash Report
Process:               BambuStudio [86484]
Path:                  /Applications/BambuStudio-SF.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.04.00.17 (???)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-12-30 12:31:56.1864 +1300
OS Version:            macOS 12.6.1 (21G217)
Report Version:        12
Anonymous UUID:        D2A2C85C-5154-CBFF-0E39-5607773E3B1C

Time Awake Since Boot: 350000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [86484]

VM Region Info: 0 is not in any region.  Bytes before following region: 4391002112
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      105b96000-1091de000    [ 54.3M] r-x/r-x SM=COW  ...S/BambuStudio

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   BambuStudio                   	       0x105f9e5bc Slic3r::ModelObject::get_instance_max_z(unsigned long) const + 188
1   BambuStudio                   	       0x105f93e20 Slic3r::ModelObject::ensure_on_bed(bool) + 416
2   BambuStudio                   	       0x10691cf83 Slic3r::GUI::Plater::changed_object(int) + 51
3   BambuStudio                   	       0x1066b1e7f Slic3r::GUI::ObjectList::reorder_volumes_and_get_selection(int, std::__1::function<bool (Slic3r::ModelVolume const*)>) + 287
4   BambuStudio                   	       0x1066bb8f6 Slic3r::GUI::ObjectList::load_mesh_part(Slic3r::TriangleMesh const&, wxString const&, bool) + 726
5   BambuStudio                   	       0x10675e55f Slic3r::GUI::GLGizmoText::on_render_input_window(float, float, float) + 3551
6   BambuStudio                   	       0x106700882 Slic3r::GUI::GLGizmoBase::render_input_window(float, float, float) + 18
7   BambuStudio                   	       0x10676b633 Slic3r::GUI::GLGizmosManager::do_render_overlay() const + 1859
8   BambuStudio                   	       0x1065d9d7d Slic3r::GUI::GLCanvas3D::_render_overlays() + 365
9   BambuStudio                   	       0x1065d6ce6 Slic3r::GUI::GLCanvas3D::render(bool) + 2358
10  BambuStudio                   	       0x1065e36bd Slic3r::GUI::GLCanvas3D::on_mouse(wxMouseEvent&) + 301
11  BambuStudio                   	       0x10883721e wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
12  BambuStudio                   	       0x108836f8b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
13  BambuStudio                   	       0x108836ea0 wxEvtHandler::ProcessEvent(wxEvent&) + 96
14  BambuStudio                   	       0x1088373bc wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
15  BambuStudio                   	       0x108777589 wxWidgetCocoaImpl::DoHandleMouseEvent(NSEvent*) + 89
16  BambuStudio                   	       0x108771da0 wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 224
17  AppKit                        	    0x7ff80e46adf8 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2704
18  AppKit                        	    0x7ff80e46a14e -[NSWindow(NSEventRouting) sendEvent:] + 352
19  BambuStudio                   	       0x10875e70b -[wxNSWindow sendEvent:] + 123
20  AppKit                        	    0x7ff80e468524 -[NSApplication(NSEvent) sendEvent:] + 352
21  BambuStudio                   	       0x1086dc842 -[wxNSApplication sendEvent:] + 98
22  AppKit                        	    0x7ff80e72118b -[NSApplication _handleEvent:] + 65
23  AppKit                        	    0x7ff80e2e9d3e -[NSApplication run] + 623
24  BambuStudio                   	       0x1087551de wxGUIEventLoop::OSXDoRun() + 174
25  BambuStudio                   	       0x108841a8d wxCFEventLoop::DoRun() + 29
26  BambuStudio                   	       0x1087b63e5 wxEventLoopBase::Run() + 85
27  BambuStudio                   	       0x10878e8f4 wxAppConsoleBase::MainLoop() + 116
28  BambuStudio                   	       0x10871726a wxApp::OnRun() + 26
29  BambuStudio                   	       0x1087ea1c3 wxEntry(int&, wchar_t**) + 131
30  BambuStudio                   	       0x1066a8af7 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 167
31  BambuStudio                   	       0x105ba6c44 Slic3r::CLI::run(int, char**) + 18324
32  BambuStudio                   	       0x105bc1310 main + 256
33  dyld                          	       0x1178ad52e start + 462

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	    0x7ff80b7bd97a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80b7bdce8 mach_msg + 56
2   CoreFoundation                	    0x7ff80b8c136d __CFRunLoopServiceMachPort + 319
3   CoreFoundation                	    0x7ff80b8bf9f8 __CFRunLoopRun + 1276
4   CoreFoundation                	    0x7ff80b8bee3c CFRunLoopRunSpecific + 562
5   AppKit                        	    0x7ff80e4669ce _NSEventThread + 132
6   libsystem_pthread.dylib       	    0x7ff80b7fa4e1 _pthread_start + 125
7   libsystem_pthread.dylib       	    0x7ff80b7f5f6b thread_start + 15

SoftFever, 2023-01-01:
Will be fixed in 1.4.1.

closed 2023-01-01 by SoftFever

Tonyknz, 2023-01-01:
Thanks,Keep it up, you are doing such a great job with this and all of your effort is really appreciated.Many thanksTonySent from my iPad

This email has been
 filtered by SMX. For more information visit
 smxemail.com