OrcaSlicer/OrcaSlicer #78: M1 Mac - Error when opening the program

DIYLAB-DE, 2022-12-30:
**Describe the bug**
When opening the program on a M1 Mac, I get a message that the program is corrupted and cannot be run. See screenshot.

**Screenshot**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.0.1 Ventura on M1 Mac

SoftFever, 2022-12-30:
Please check NOTE 1 part in the release notes: https://github.com/SoftFever/BambuStudio-SoftFever/releases/tag/v1.4.0

DIYLAB-DE, 2022-12-30:
**Very nice, it works, thank you!**
Is there a possibility later to sign this program properly?

SoftFever, 2022-12-30:
Sure.
I will do that when there are more Mac users.

closed 2022-12-30 by SoftFever