OrcaSlicer/OrcaSlicer #83: Bed Shape Not Saved

ShryuKt, 2022-12-31:
Hi, 

Just a bug I found.
Even if I select a Texture and Model for the Bed Shape, It is not saved.

Select
[image]

OK and reopen
[image]

And of course the texture and the model doesn't change.

Thanks
-- Shryû --

Arakon, 2023-01-28:
This issue still occurs. Is there a workaround? I looked around in the config json files, but there's not even a mention of these settings that I could find.

Icesythe7, 2023-03-28:
> This issue still occurs. Is there a workaround? I looked around in the config json files, but there's not even a mention of these settings that I could find.

Go to `C:\Users\Icesythe7\AppData\Roaming\OrcaSlicer\user\{random number here}\machine` open your printer.json file
insert on line 2 and 3
```
"bed_custom_model": "C://Users//pathtoyourstl//printbed.stl",
"bed_custom_texture": "C://Users//pathtoyourpng//Rook_Grid.png",
```

then save and restart
The error seems to be that when saving the texture/model it is omitting the "//" in the path thus making the path invalid and it gets removed.

SoftFever, 2023-04-18:
Fixed in 1.6.2 beta

closed 2023-04-18 by SoftFever