OrcaSlicer/OrcaSlicer #88: Feature request to add calibration lines to pre sliced build plate

keenzkustoms, 2023-01-01:
I have created this svg and stl for the bambulab plate to show the calibration lines before slicing to help you position objects on the plate before you slice it to prevent an overlap in the first layer. 
[image]

https://www.printables.com/model/321899-bambulab-slicer-bed-texture-with-calibration-marke

Would be good it we could get like a toggle switch to turn it on and off. 
Cheers

SoftFever, 2023-01-01:
I like your idea and work!
I will add this in the next release and credit you.

keenzkustoms, 2023-02-28:
@SoftFever any update on this? cheers

SoftFever, 2023-04-05:
@keenzkustoms Done! Thanks for creating the texture!

closed 2023-04-05 by SoftFever

Learoy1, 2023-04-06:
> I have created this svg and stl for the bambulab plate to show the calibration lines before slicing to help you position objects on the plate before you slice it to prevent an overlap in the first layer. [image]
> 

oh wow you used my model in the photo 🥇

keenzkustoms, 2023-04-06:
@learoy1 yes I did thankyou! Shows it really well how useful it is when trying to line up your part with the calibration lines 😀

CoryCharlton, 2023-04-08:
I noticed that when this feature is enabled in 1.6.0+ the plate name is replaced with the black cross hatch. Not sure if this is the intended behavior or if anything could be done about it but I figured I'd mention it. Let me know if you'd prefer I opened a new issue

keenzkustoms, 2023-04-09:
@CoryCharlton yea might be worth making a separate bug report