OrcaSlicer/OrcaSlicer #89: Bambu Studios Pluging Download Failed

Tyler8299, 2023-01-01:
Program Crashes about 3x before it launches. when it does it fails to download Bambu Plugin. I disabled all Windows antivirus still did not work. was able to continue using the last version without issues. thank you and happy new year and thank you for everything

[image]

SoftFever, 2023-01-01:
@Tyler8299 I have no idea why it failed to download plugin on some machines.
As a temporary workaround, You can try to copy %appdata%\BambuStudio\plugins folder to %appdata%\BambuStudio-SoftFever\plugins manually.

davidja92, 2023-01-01:
Mine was failing and i force quit microsoft teams and then the plugin downloaded. i do not know the correlation but i can replicate it every time

Tyler8299, 2023-01-01:
force quitting teams did the trick thank you!!

Tyler8299, 2023-01-01:
BTW i have a Creality CR10V3 and Ender 3 S1 profile preset for Bambu Labs
works amazing.

hliebscher, 2023-01-02:
can you check in CURL.cmake

if (WIN32)
  #set(_curl_platform_flags  ${_curl_platform_flags} -DCMAKE_USE_SCHANNEL=ON)
  set(_curl_platform_flags  ${_curl_platform_flags} -DCMAKE_USE_OPENSSL=ON -DCURL_CA_PATH:STRING=none)

edit pls and recompile ... i can't check no windows

that is the difference to org Bambu

Tyler8299, 2023-01-02:
I do not know how to do this

Tyler8299, 2023-01-02:
I know you are a busy person, id be willing to let you rummage around on my machine using a remote desktop to help your Software if we are both available. Thank you again your Software is amazing

davidja92, 2023-01-06:
@hliebscher this issue is also persistent with the normal bambu studio closing teams still fixes it.

WareWolfSoftworks, 2023-01-27:
For anyone else with this issue I had already uninstalled Teams. I force closed the Edge program that was running in the background. I don't use Edge but apparently it starts a service no matter what. So if you don't have Teams give that a shot.

SoftFever, 2023-05-06:
Closing as problem solved.

closed 2023-05-06 by SoftFever