OrcaSlicer/OrcaSlicer #93: Flow Calibration function does not respect max volumetric flow rate

psiberfunk, 2023-01-02:
When doing the flow calibration workflow, I was looking closely and realized that the max _volumetric_ flow rate set by the filament was getting exceeded, potentially leading to bad/misleading results as a function of exceeding the maximum flowrate the hotend can handle with the filament.    Instead of scaling the up and down around the maximum volumetric flow rate that the filament specifies to run the test, the MAXIMUM flow test should, at worst hit the volumetric cap, and everything else should be less.

Here's an example where I set the volumetric cap to 18, but you can see that the +20% blocks is far exceeding this rate:  
[image]

The 0% block hews exactly to my limit.. but this is likely incorrect.  Speeds should be scaled so that all volumetric rates are at or below the global limit the filament sets.

This should lead to better filament calibrations and less false readings merely because you are exceeding the global volumetric flow rate.

psiberfunk, 2023-01-02:
Having thought about this some more, this is actually a slightly more complex issue.  The key here is you want to be sure you are always within the volumetric flow cap , BUT you also want to ensure other things that can affect the surface calibration quality, for example, printing the lines at the same speed.. as lines printed at different speeds often have different surface finish qualities that might mislead the user.

The safest thing to do here might be to scale the surface, infill, and wall speeds down so that you know you are always inside the volumetric cap.    Basically, detect if any of the max flow rate modifier would put you over the filament's volumetric cap.  If it would, then scale speeds down equally for ALL of the objects so that the fastest speed combined with the highest-flow-rate object would not exceed the cap.

In this way, you can ensure that the maximum flow rate cap is not exceeded AND that the print speeds are uniform across all test patches, which is important for comparative quality.

It means that the pattern won't print as fast as the flow cap rate might technically allow, but it makes sure that the test results stay valid, so long as the volumetric flow rate is valid.   It might be wise to also advise the user to make sure they are working within the sane limits of the volumetric flow rate when running this pattern.   It is much better to calibrate flow in the "linear" extrusion region of the extrusion amount vs speed graph than to get into the non-linear region.. because then things WILL get weird and the calibration will not be reliable as slowing down will actually increase extrusion efficiency.

From : https://www.cnckitchen.com/blog/flow-rate-benchmarking-of-a-hotend  , you want to stay outside the region where you start to spike in under-extrusion.  For example, in this particular graph, going above 12 mm^3/sec for this filament/hotend/temp is going to be lead to confusing "calibration" results... as when you slow down the volumetric speed, you will actually extrude more filament for a given set extruder ratio!

[image]

SoftFever, 2023-01-03:
Thanks for bringing this up. 
Indeed all blocks should respect the volumetric speed limitation.

I struggled with deciding on the appropriate volumetric speed limit to use while implementing this feature. 
Initially, I set the limit to 9 for flow rate calibration, but after conducting several test prints, I realized that it would be more beneficial to have calibration settings that closely mimic the actual printing settings

SoftFever, 2023-01-05:
fixed

closed 2023-01-05 by SoftFever