OrcaSlicer/OrcaSlicer #94: PA (lines) Calibration function does not respect max volumetric flow rate

psiberfunk, 2023-01-02:
Experimenting with using the PA calibration function (with lines), I setup a filament as with bug #93 , and again, I noticed behavior that was exceeding the volumetric flow rate cap the filament has set, which will lead to misleading PA results, as a normal print WILL respect this flow rate cap.

Here, with an intentionally low volumetric rate cap of 2 mm^3/sec, I notice that the calibration setup is blindly setting the speed to the outer wall speed instead of seeing if doing so would hit the maximum volumetric flow rate at 0.2 mm layer height.  I therefore am seeing lines extruded way above the volumetric flow rate cap of 2 mm^3/sec, in this screenshot, at 24 mm^3/sec !  

Of course this is no good, as the PA results can't be trusted if they're exceeding the hotend/extrusion capacity.  The test should determine the "fast" line speed by adjusting the outer-wall speed down if necessary to stay within the volumetric flow rate cap.  This way the test should remain sane and useful to the user (assuming they've set a sane volumetric flow cap).

[image]

It is totally reasonable, by the way, for a user to have an outer-wall-speed set faster than the volumetric flow rate cap will allow.  This is why the volumetric flow rate can be used to control for limitations in filament extrusion and still use the same process profile as another.

psiberfunk, 2023-01-02:
This is a dupe of #91 I just realized.  Probably can close both at the same time when a fix is made.

SoftFever, 2023-01-03:
Ya, it should respect the volumetric speed limitation

SoftFever, 2023-01-05:
Fixed.

closed 2023-01-05 by SoftFever