OrcaSlicer/OrcaSlicer #99: Adjust PA calibration range

steven850, 2023-01-02:
We need to be able to adjust the range of the PA calibration values.  a max of 0.096 is way too low for TPU. We need a way to set a custom start and end value for the test.

psiberfunk, 2023-01-03:
I agree .. a pop up with min/max/step size would be useful .. with defaults going to the existing numbers of course

psiberfunk, 2023-01-03:
PS make sure validate that the pattern will fit on the bed in case the user sets unreasonable values like step size of 0.00001

steven850, 2023-01-03:
Ideally also a way to adjust the min and max speed, so for some materials there limited to 30mm/s, so it going from 20 to 30 doesn't provide much of a difference in speed for a good comparison, so being able to drop the current 20mm/s to like 5, so having it go 5mm to 30mm/s would provide a better calibration for that material.

psiberfunk, 2023-01-03:
You can already do that by setting the outer wall speed @steven850 .  Problem is , doesn’t respect volumetric flow rate limits , yet .. but that’s getting fixed in #94

steven850, 2023-01-04:
@psiberfunk Yes but my point is, if I set that to 20mm/s, the test is just a line at a constant speed, there is then no speed change. need to be able to change the speed of the "slow part" of the test aswell. So the line will do for example 5 slow 20 fast.

psiberfunk, 2023-01-04:
> @psiberfunk Yes but my point is, if I set that to 20mm/s, the test is just a line at a constant speed, there is then no speed change. need to be able to change the speed of the "slow part" of the test aswell. So the line will do for example 5 slow 20 fast.

I guess that’s true .. I’m trying to think of a case where 20 isn’t “slow” though … :)

steven850, 2023-01-05:
When the max speed of the filament is 20 or 30, in that case need to drop down to 3 or 5mm/s  I have some TPU that wont do more then 2 cubic mm/s, so max print speed on that is like 20, hence needing these settings.

psiberfunk, 2023-01-08:
> Ideally also a way to adjust the min and max speed, so for some materials there limited to 30mm/s, so it going from 20 to 30 doesn't provide much of a difference in speed for a good comparison, so [...]

Can I suggest maybe that you make this a separate ticket as it's arguably a separate feature/knob to turn ?

bani6809, 2023-01-08:
starting value and step value would be useful. 0.098 seems to be the very bottom of Overture 95A TPU.

closed 2023-02-10 by steven850