OrcaSlicer/OrcaSlicer #102: Line width for Support Interface and for Overhangs

rsilvers129, 2023-01-03:
If you add line width for Support Interface and also for Overhangs (or anything over support interface) then we can make those lines thinner, such as 0.3mm thick. That will make support peel off easier which will allow for a 0.1mm Top Z Distance. I tested this and it works. For that matter, a way to have the fan(s) come on a certain percent when doing any area that support interface contacts the part so that it doesn't stick as well.

SoftFever, 2023-01-03:
Could you share some more information about the settings?
Some photos or a sample project will be beneficial.

We already have the `Top interface spacing`  setting to achieve a similar goal unless I missed something here.

rsilvers129, 2023-02-21:
Not interface spacing and not support line width. Just Support "Support Interface line width" and "Overhang Line Width."

I promise this will be a major breakthrough.

If you print support interface and overhangs with a 0.30 to 0.37mm line width, the support will peel off, even if there is only a 0.1mm support gap. Problem is, the slicer has no way to set the support interface line width.

For now, the best I can simulate it is to set the Internal Solid Infill and Inner Wall line width to 0.35mm to come close to this. It makes support come off as easily with a 0.1mm gap as it currently does with normal settings and a 0.2mm gap. I have tested it.

[image]

Bonaparte1446, 2023-07-26:
that could be very useful also using material support, hope that could be add. thank

closed 2023-11-25 by github-actions[bot]