OrcaSlicer/OrcaSlicer #103: Object Flow Ratio Does Not Support Decimals

CaZa3D, 2023-01-03:
**Describe the bug**
When changing flow ratio under specific objects to decimals  such as 95.5%, they revert back to hole numbers such as 95%

**To Reproduce**
Steps to reproduce the behavior:
1. Click Objects
2. select object
3. Quality tab
4. Set flow ratio to 95.5%
5. click away
6. click back on object
7. flow ratio is now at 95%

**Expected behavior**
I expect to type 95.5% and it to stay at 95.5%

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 x64
 - Version Softfever  1.4.1

hliebscher, 2023-01-03:
at osx 13.1 is the same
does it make sense? the deviation from the filament is greater

SoftFever, 2023-01-03:
It's a UI bug. It will be fixed in the next update.

SoftFever, 2023-01-03:
> 

I guess it depends on the base flow ratio set in the filament setting.
If that is set to 1, then 1% is too big

SoftFever, 2023-01-05:
Fixed.
Will be supported in next release

closed 2023-01-05 by SoftFever