OrcaSlicer/OrcaSlicer #105: GCode error with Pressure Advance Test Tower DDE during print

hliebscher, 2023-01-03:
**Error: Fan number 2 not found**

M572 D0 S0.05; Override pressure advance value
M106 S0 ; disable fan
**M106 P2 S0 ; disable additional fan**  <<-- it is only for BBS Printer 

i think its in the file print.hpp
 
   //SoftFever
    bool &is_BBL_printer() { return m_isBBLPrinter; }
    const bool is_BBL_printer() const { return m_isBBLPrinter; }
    CalibMode& calib_mode() { return m_calib_mode; }
    const CalibMode calib_mode() const { return m_calib_mode; }

SoftFever, 2023-01-03:
I keep that option available to all printers on purpose. 
Nowadays, the auxiliary fan is not rare among DIY printers.
Users can define a macro to enable this command.
(not sure if RRF can override a builtin command, Klipper can)

SoftFever, 2023-01-03:
Anyway, unticking this option in the printer settings should solve your problem in case you don't have an auxiliary fan.
[image]

hliebscher, 2023-01-03:
it is in all gcodes 
the aux is not enabled 

; filament start gcode
M572 D0 S0.025; Override pressure advance value
M106 S0 ; disable fan
M106 P2 S0 ; disable additional fan 
;LAYER_CHANGE

hliebscher, 2023-01-03:
[image]

SoftFever, 2023-01-03:
Noted.
It's bug then

SoftFever, 2023-01-05:
Fixed.

closed 2023-01-05 by SoftFever