OrcaSlicer/OrcaSlicer #108: don't select TRONXY printer it crashes tha app on MAC !!!

hliebscher, 2023-01-03:
after selecting Tronxy it crashes 
is not possible to restart it crashes again

only kill the config directory helps...

Failed loading configuration file /Users/hliebscher/Library/Application Support/BambuStudio-SoftFever/system/Tronxy/machine/Tronxy X5SA 400 0.4 nozzle.json
Suggest cleaning the directory /Users/hliebscher/Library/Application Support/BambuStudio-SoftFever/system firstly

it is reproducible 2 times

SoftFever, 2023-01-04:
https://github.com/SoftFever/BambuStudio-SoftFever/pull/81#issuecomment-1370355732

erebus04, 2023-01-04:
I'm looking at this on my Mac. Which MacOS version are you using?

SoftFever, 2023-01-04:
@erebus04 I believe this happens on all OS.
You can reproduce it by remove `~/Library/Application Support/BambuStudio-SoftFever/system` folder then run the application again.

erebus04, 2023-01-04:
Ok thanks. I Check all

closed 2023-01-05 by hliebscher