OrcaSlicer/OrcaSlicer #110: First Layer printing indicator

cdmichaelb, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
While the first layer is printing, have some kind of indicator.

**Describe alternatives you've considered**
AI Failure detection, however it does not prevent failures.

**Additional context**
I would like to print my prints at manual 50% speed for first layer, even with the defaults set to lower speeds I still have some failures. However I have no way of knowing if it's currently printing first layer, or additional layers.

Thanks.

cdmichaelb, 2023-01-03:
I ended up just lowering the default, but I still think it would be nice to know if first layer has succeeded.

SoftFever, 2023-01-04:
I also would like to see this feature.
Unfortunately, there is not much we can do here(forked slicer) ourselves. 
It is entirely up to Bambu to add features in their firmware."

closed 2023-01-04 by SoftFever