OrcaSlicer/OrcaSlicer #114: Travel moves are missing feed rate, resulting in very slow travel moves

rlabusiness, 2023-01-04:
I've been really excited to discover this SoftFever fork of Bambu Studio, and I'd love to make it my one and only slicer. Thanks for all the work put into this!

I'm experiencing really slow travel moves on my CR-10 Smart Pro and have confirmed a potential fix. I'm hoping this can be resolved in the slicer. 

As you can see in the attached screenshot, travel moves in the GCODE generated here don't explicitly include the feedrate (F9000). When I add these to the file manually, travel moves are as expected (fast).

I made a test printer profile for my CR-10 Smart Pro starting from the CR-10 Max profile, then changed its dimensions to match my printer and GCODE flavor to Klipper. I have a Creality SonicPad which runs Klipper and Moonraker behind the scenes (versions listed below).

SonicPad Firmware: 1.0.6.35.154 (latest)
Klipper firmware: 0.1.27-0-gd8759023-dirty
Moonraker: 0.0.56-0-g86afbaf-dirty

[image]

SoftFever, 2023-01-05:
@rlabusiness 
The g-code is fine.
The firmware should use the previously set speed(F9000).
[image]
Can you check if you have arc support enabled in your Klipper config?
`
[gcode_arcs]
resolution: 0.1
`
I suspect you don't have ARC enabled hence Klipper skipped the G3 command, which result in using F1800 speed for the following travel move.

You can also change `Z hop type` to `NormalLift` in the extruder setting if you don't plan to use any ARC move commands anyway.
[image]
 This is should also fix your problem.

rlabusiness, 2023-01-05:
@SoftFever Wonderful! This solved it. I really appreciate the explanation as well. Again, it's very appreciated what you're doing here.

closed 2023-01-05 by rlabusiness