OrcaSlicer/OrcaSlicer #115: Option to NOT Cut Filament In AMS After Print Is Finished

djngnear, 2023-01-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I would love to have an option to keep filament loaded in the hotend after a print.  A lot of times I'd like to use the same loaded filament after a print only to have it automatically cut and unloaded.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Maybe have the setting on the Print Plate dialog box when sending it to the printer?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I don't know a work around to keep it loaded in the hotend.

**Additional context**
Add any other context or screenshots about the feature request here.

steven850, 2023-01-05:
You can easily do that in your printer profile. Open up the profile you want, like the 0.2mm standard profile. Then go to machine end code. And look for the following

> G1 X100 F12000 ; wipe
> ; pull back filament to AMS
> M620 S255
> G1 X20 Y50 F12000
> G1 Y-3
> T255
> G1 X65 F12000
> G1 Y265
> G1 X100 F12000 ; wipe
> M621 S255
> M104 S0 ; turn off hotend

simply delete from  M620 up to M104 so it looks like this  

> G1 X100 F12000 ; wipe
> ; pull back filament to AMS
> M104 S0 ; turn off hotend

now save the profile as 0.2mm Leave filament or something like that. So whenever you print with that profile it will leave the filament loaded at the end of the print.

djngnear, 2023-01-06:
Thank you! I will definitely do that. I'd still like the option...

closed 2023-05-06 by SoftFever