OrcaSlicer/OrcaSlicer #117: Vase mode does not work

rollduck, 2023-01-05:
**Describe the bug**
Afert vase mode selection an error occurs:
"Unhandled unknown exception; terminating the application"
06:44:36: BambuStudio got an unhandled exception: boost::bad_any_cast: failed conversion using boost::any_cast
06:44:36: BambuStudio got an unhandled exception: boost::bad_any_cast: failed conversion using boost::any_cast

**3mf File for This Bug**
No matter what object we would like to slice. 

**To Reproduce**
Steps to reproduce the behavior:
1. Just select and confirm spiral vase

**Expected behavior**
I expected a sliced model :) 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 
 - Version 22H2
 - SoftFever version 1.4.1

hliebscher, 2023-01-05:
it is the same at MAC M1 but not with BBS Printer

and it works if you **set before** walls to 1 and top to zero

i have repeat this 3 times

SoftFever, 2023-01-05:
Thanks, @rollduck, and @hliebscher .
The fix will be included in the next release

closed 2023-01-05 by SoftFever