OrcaSlicer/OrcaSlicer #118: Add Surface Mode from Cura

gfalgiano, 2023-01-05:
**Is your feature request related to a problem? Please describe.**
I'm trying to print a model from Planeprint.com and they require Cura just for this one feature, surface mode. Without surface mode, I can't get the Bambu Studio to print the internal structure of the model. https://www.planeprint.com/print

**Describe the solution you'd like**
If possible, it would be great to have support for Surface Mode as described here: https://support.makerbot.com/s/article/1667418049490

**Describe alternatives you've considered**
 I see that Bambu Studio has picked up "Even-Odd" mode from Prusa slicer to support 3D printed planes from another designer 3D Lab Print https://3dlabprint.com/faq/prusaslicer/ , but the slicer settings don't work for PlanePrint files.
 
**Additional context**
Similar request for Prusa slicer: https://github.com/prusa3d/PrusaSlicer/issues/6982
One user suggests this feature can benefit academic research into 3D printing tissue for medical applications. 

Here are some images of a sample file comparing the model vs sliced results. You can see that the internal structure is missing.
[image]
[image]

psiberfunk, 2023-01-08:
Personal opinion :  This is probably a significant amount of work to implement and may not really be on-brand for this slicer fork.  I'd advocate for something like this in SuperSlicer/PrusaSlicer, as I think they're more likely to have the bandwidth and experience to implement something like this from Cura.

Thoemse, 2023-03-04:
It is also the single most useful feature to print light weight parts though. That alone should make it worth the effort.

psiberfunk, 2023-03-04:
Sure , I didn’t say it wasn’t worth the effort , I was just saying this might not be the right (single person) team to advocate for doing the work .  I think you might have more luck upstream

closed 2023-11-25 by github-actions[bot]

hwinkel, 2023-11-25:
can this really be closed, after nothing happened?

Grinderman77, 2024-03-22:
Unfortunately I have no programming background, but I wanted to bump this up anyway. The possibilities are endless, especially if you want to define the location of suppoert structure in 3D printed planes. Although there has been quite a proliferation of planes that use infill as internal structure which is more or less a lazy shortcut for the designer compared to the former, it would be very helpful if we could support those designers who go the extra mile.

dewi-ny-je, 2024-07-14:
> Unfortunately I have no programming background, but I wanted to bump this up anyway. The possibilities are endless, especially if you want to define the location of suppoert structure in 3D printed [...]

Or simply open the same ticket on PrusaSlicer