OrcaSlicer/OrcaSlicer #119: Request: Toolchange gcode

Prutsium, 2023-01-05:
SuperSlicer has a feature we heavy use for Multimaterial (single extruder) printing called Toolchange Gcode under printer settings.
Would it be possible to enable this too in in your version?

[image]

SoftFever, 2023-01-05:
It's already there.
It called `Change filament G-code`
[image]

SoftFever, 2023-01-05:
Yeah, a lot of variable names have been changed.
I'm starting to add more back-compatible variable names to the fork slowly.
So people can port custom g-code without many hassles.

SoftFever, 2023-01-05:
Anyway, I'm closing this issue as this is not relevant to the original issue anymore.

closed 2023-01-05 by SoftFever