OrcaSlicer/OrcaSlicer #120: Calibration on MacOS

chillee151, 2023-01-05:
In the calibration wiki article, the calibration tab is above the main window on a PC. Unfortunately, I see no calibration option at all on a mac. Is there a way to enable it? Perhaps I'm misunderstanding the feature, but I can't find it anywhere. 
[image]

SoftFever, 2023-01-05:
@chillee151 it's on the global  menu bar(the one has Apple menu)

urbankuest, 2023-01-06:
@chillee151 are you on V1.4.1, works fine on my Mac M1

[image]

TheMediaDude, 2023-01-06:
Same issue here and I'm on V1.4.1.
[image]

[image]

TheMediaDude, 2023-01-06:
Works for me after a restart.

Manfredo1952, 2023-01-06:
Indeed, I had to close and open the App 2 times or so, thereafter the calibration tab appeared (iMac 11.7.2)

chillee151, 2023-01-06:
Indeed it showed up after a restart. Thanks for the help. Not sure why it didn't appear at first.

antonme, 2023-01-06:
It seems that Calibration menu disappears if you log in to your Bambu Labs account (and if you were logged in already, it disappears right at the app start, in a few seconds)

Manfredo1952, 2023-01-07:
> It seems that Calibration menu disappears if you log in to your Bambu Labs account (and if you were logged in already, it disappears right at the app start, in a few seconds)

I have an iMac here. When I start the app, I am already logged in and the calibration menu appears **and stays**. When I log out, yes, then it disappears and does not come back when I log in again. I close and reopen the app and it is back.

SoftFever, 2023-01-07:
Thanks for all the reports and workaround.
Let me check this part of the codes again to see if I can find anything.

Everything works fine on my Mac...

mrskoku, 2023-01-07:
> 

macbook air m1, macos monterey 12.6.2 - calibration menu was missing (it's fine after restart)
macbook pro 13 (a1502, intel i5) - big sur 11.7- calibration menu is always visible

antonme, 2023-01-08:
@SoftFever macbook16 2019 (intel i9), 13.1 Beta (22C5050e). For me, Calibration menu always visible when logged out of Bambu account (after app restart) and always invisible when logged in (immediately)

mrskoku, 2023-01-08:
> @SoftFever macbook16 2019 (intel i9), 13.1 Beta (22C5050e). For me, Calibration menu always visible when logged out of Bambu account (after app restart) and always invisible when logged in (immediately)

thank you. i was able to replicate this.

softfever 14.2

app started with account logged in - calibration menu disappear 
i've logged off in app
app closed
app started with account not logged in - calibration menu is visible
i've logged in to bambu account- calibration menu disappear immediately
app closed
app started with account logged in - calibration menu is visible

Manfredo1952, 2023-01-08:
I have an iMac here (15,1 - Intel with OSX 11.7.3) - my experience is different 

- app started with account logged in - calibration menu appears and will not disappear
- i've logged off in app  - calibration menu disappears .but app stays alive, with account logged off
- app started with account not logged in - calibration menu is visible
- i've logged in to bambu account- calibration menu disappears immediately indeed)
- app closed
- app started with account logged in - calibration menu is visible

Do you have an Intel or ARM type of Mac (I assume you have installed the correct version of this software?)

antonme, 2023-01-08:
@Manfredo1952 intel. But I may have found out what was causing the difference: calibration menu only disappears on start when profile sync is turned **off** for Bambu account. When it is on, restarting the app helps indeed.

SoftFever, 2023-01-15:
Hi guys,

Can you try this latest build? 
I pushed a fix, it should fix this issue. 
Need someone to test for me:
Intel: https://github.com/SoftFever/BambuStudio-SoftFever/actions/runs/3921402505
Apple silicone: https://github.com/SoftFever/BambuStudio-SoftFever/actions/runs/3921402501

chillee151, 2023-01-15:
The fix seems to work for me on an M1! THank you @SoftFever

Manfredo1952, 2023-01-15:
Just tried the Intel version. I can start the program and it does log in with calibration menu active. I can log out and log in again, it does stay. BUT   on first start, there is a help menu, which disappears when I log out and does not come back when I log in again. Only when I end and start again, the help menu comes back. 
[image]

antonme, 2023-01-16:
@SoftFever:
> Can you try this latest build?

Everything seem to work fine for me with that new build. Thank you!

Alsidy, 2023-04-23:
I have a 2016 MacBook Pro, Intel running OS X 13.3. I've just run a calibration test twice and the filament profile disappears when I close Orca. the second time I did the profile, I made double sure I saved it, made sure it appeared in user profile and closed, reopened and it's not there. I'm trying to create a profile for a Creality Printer if that makes any difference? Do I have to sign in on my Bambu profile?

shaneshort, 2023-09-13:
@Alsidy this is probably worth opening a separate issue for. I might suggest having a look in the following path to see if the filament file is actually being saved, though:
/Users/yourusernamehere/Library/Application Support/OrcaSlicer/user/default/filament

Jo-Con-El, 2023-12-05:
> Just tried the Intel version. I can start the program and it does log in with calibration menu active. I can log out and log in again, it does stay. BUT on first start, there is a help menu, which [...]

Hey, @Manfredo1952, it's been almost a year since you reported problems with the disappearing menus.  Is this still happening?

closed 2024-03-12 by github-actions[bot]