OrcaSlicer/OrcaSlicer #127: Wrong extrusion if detect overhangs walls is active

abstractFlo, 2023-01-07:
**Describe the bug**
Currently i print some parts for an MK3S Bear Upgrade. If i enable "Detect Overhang Walls" then there is a false extrusion near an overhang

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Open 3mf File and slice
2. Look at layer 79 in preview
3. Look at overhang
4. Go to Quality -> remove "Detect overhang walls"
5. Slice and look. The line is gone

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10
 - Latest build for bambu studio fork

SoftFever, 2023-01-07:
Thanks.
I'm able to repro this issue.

hliebscher, 2023-01-07:
the bug is only with classic arachne works at mac

hliebscher, 2023-01-07:
crazy if you go back from arachne to classic then it works without the bug...
[image]

SoftFever, 2023-01-07:
I can reproduce this bug with BambuStudio 1.4.0.18
It's an upstream bug.
The key to reproducing this is setting the `default` and `Outer wall` width to 0.4.

I'm going to open to the ticket in their repo. They probably can fix it fast.

SoftFever, 2023-01-07:
@hliebscher this is most likely because when we switch back to classic from Arachne, the slow down for overhangs remain unchecked :(

SoftFever, 2023-01-13:
Fixed in b10799027012a6c530da78987e5d416f8fcfd5c4
Nightly build are available now: https://github.com/SoftFever/BambuStudio-SoftFever/actions

closed 2023-11-25 by github-actions[bot]