OrcaSlicer/OrcaSlicer #128: after load 3mf the api key (id addr) was not load from conf file

hliebscher, 2023-01-07:
after load 3mf the api key (id addr) was not load from conf file
[image]

only if you reload the orange button in printer config
[image]
[image]

SoftFever, 2023-01-07:
Ah, yes, this is an issue I wanted to fix but always forgot to fix.
Let me see if it's an easy fix
If so, I will squeeze this fix into the release.

hliebscher, 2023-01-07:
super ! very cool

closed 2023-01-07 by hliebscher