OrcaSlicer/OrcaSlicer #134: Patterntype for Solid and top layer

TerrorzwergOlli, 2023-01-07:
Would be nice to implement something like SuperSlicer did:  The option to set line not to be connected to each other.
[image]

This realy  helps with top finish. Espacally in small areas.
Result looks like this:
[image]

ShryuKt, 2023-01-08:
Hi,

I'm a big fan of those options
I already ask for them in #50

pemerick07, 2023-01-08:
This is very helpful when trying to print strong dense parts that aren't over extruded at the interface between infill and perimeters.

SoftFever, 2023-01-09:
You can achieve the same effect by changing `Top surface pattern` to `Monotonic line`
[image]

For comparison, this is `Monotonic`
[image]

TerrorzwergOlli, 2023-01-09:
> You can achieve the same effect by changing `Top surface pattern` to `Monotonic line` [image]
> 
> For comparison, this is `Monotonic` [image]

I tried  this before (I thought maybe this is the option) but it haven't change anything
[image]

TerrorzwergOlli, 2023-01-09:
funny enough: in the x1 profile it works

TerrorzwergOlli, 2023-01-09:
@SoftFever I am unsing 1.4.2 and here is the project file maybe it is a bug?
[flow_test.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10369910/flow_test.zip)

closed 2023-11-25 by github-actions[bot]