OrcaSlicer/OrcaSlicer #135: Low poly prints are broken

H4kunaM4tata, 2023-01-08:
First of all, this is not your responsibility but BambuLab seems to don't know how to solve this or aren't taking this problem as serious as they should.
I have opened Git ticket, support ticket but them all got nowhere.

This ticket is a scream for help.

- https://github.com/bambulab/BambuStudio/issues/228
- https://github.com/bambulab/BambuStudio/issues/1006

X1C seems to be unable to print low poly models that we can print just fine with a MK3S+ without those quality issues.
I mean, I had mine for 2y and never had such problem, slice and print. That simple.

Again, this is a scream for help, not your responsibility.
I appreciate if you could investigate this and hopefully get things improved.

Thank you so much

psiberfunk, 2023-01-08:
There's not much @SoftFever can do here.. this is a problem with motion planning mostly, and the STEP file default imports.  Sadly it's not a slicer issue but a printer issue.  If you slice a file in a different slicer like PrusaSlicer or SuperSlicer you'll have the same problem.  The only thing to do really here is slow down on those models and/or try playing with the arc tolerance and make the resolution BIGGER (not smaller) to trick the slicer into creating less accurate, but fewer, arcs.. which will visually look better, up to a point.

SoftFever, 2023-01-08:
Do you have the model?
For this specific model, it does look like the model issue here.
Sometimes scanned models are tricky if they are not properly post-processed.

H4kunaM4tata, 2023-01-08:
@psiberfunk yeah, one of my design, I had to add line 200 vertices for X1C to print a circle instead of a "square".
Firmware change is needed indeed but it doesn't appear that BambuStudio is doing much to help.

@SoftFever to be fair, anything that has curves on it prints terrible bad. I have to either find a SLA version or a high quality design. which isn't always possible.

Don't bother man, I'll try to find a more practical way to deal with low poly prints which is the majority of the designs out there and was never a problem before. 
Let's us know if you somehow managed to get this somewhat improved

Thank you

closed 2023-01-08 by H4kunaM4tata