OrcaSlicer/OrcaSlicer #136: Prusa Slicer infill anchor length

errhoss, 2023-01-08:
I'm trying to print the Vector Finesse (https://shop.vectorfinesse.com/) open back headphones and it requires that infill anchor be set to 0 otherwise it connects the end of the infill and doesn't look or sound right. 

[image]
I'd like the "Length of the infill anchor" setting available in SoftFever. 

It will work if you slice in Prusa slicer and then print from SD card, but I can't do multi color and the quality is lacking.

This is what it looks like from Bambu studio/SoftFever
[image]

This is what it should look like with the infill anchor set to 0
[image]

errhoss, 2023-01-11:
Just to be clear I'd like access to both "length of the infill anchor" and "maximum length of the infill anchor" if possible.

millpreet, 2023-04-17:
me 3, encountered exact same issue.

closed 2023-08-13 by SoftFever