OrcaSlicer/OrcaSlicer #137: MacOS: "File > New Window" opens window of original Bambu Studio, not SoftFever version

rlabusiness, 2023-01-08:
**Describe the bug**
I have both the original Bambu Studio and the SoftFever version installed on my Mac running OS version 12.6 (Monterey). They are almost always both open at the same time. Opening a new window in SoftFever version actually opens a new window in the original Bambu Studio.

**To Reproduce**
Steps to reproduce the behavior:
1. Install both the official Bambu Studio and SoftFever versions (I installed the official one first)
2. (Optional?) Open the official Bambu Studio and start a new project
3. Open the SoftFever version and start a new project
4. In the SoftFever version, click File > New Window
5. A new instance/window of the official Bambu Studio is opened

**Expected behavior**
I would expect a new instance/window of the SoftFever version of Bambu Studio to open

**Desktop (please complete the following information):**
 - OS: MacOS 
 - Version 12.6 (Monterey)

hliebscher, 2023-01-08:
File New Window opens every time the org BambuStudio.app if installed !
New Window work not on BambuStudio-Softfever.app  
Mac M1 13.1 

@rlabusiness why do you need open the app 3 times  ?

rlabusiness, 2023-01-08:
I am often working on multiple projects destined for multiple printers simultaneously, and it disrupts my workflow to be forced to work in a single file mode.

Even with a single project, I could decide to switch filaments (for example), and I would want to print a couple of calibrations, and I find it inconvenient to have to close files I’m working on rather than simply open up a new app window.

I can’t be the only one who would appreciate this. I’m sure the feature exists in Bambu Studio and its predecessor/sibling apps because a lot of people work in a multi-threaded fashion.

Thanks!

closed 2023-05-06 by SoftFever