OrcaSlicer/OrcaSlicer #143: [FR]Fan Speedup at Overhangs/Bridging

tome9111991, 2023-01-08:
Hello,
i'm looking for a feature to increase the fan speed before a overhang or briding starts.
maybe you can set a time 2sec or 5mm for exemple. depending on what is easier to implement.

within these values ​​the fan started 2sec or 5mm before.

psiberfunk, 2023-01-09:
What you're looking for here is really the fan spin-up / prespin setting from SuperSlicer I think:   https://github.com/supermerill/SuperSlicer/issues/157  
**printer -> fan startup delay**

psiberfunk, 2023-01-09:
e.g. this would ideally be ported from there, and so @SoftFever doesn't have to re-write it from scratch.

tome9111991, 2023-01-09:
yes works nice in susl

leofaoro, 2023-01-10:
I raised this issue in the BS github: https://github.com/bambulab/BambuStudio/issues/978

closed 2023-05-06 by tome9111991